jupytercad/JupyterCAD #46: Delete key binding

martinRenou, 2022-12-05:
Add a `Delete` key binding for removing shapes

arjxn-py, 2024-07-24:
Hi, I did some workaround to try having a (`Ctrl` + `x`) key binding for this - 
```javascript
  componentDidMount() {
    document.addEventListener('keydown', this.handleKeyDown);
  }

  componentWillUnmount() {
    document.removeEventListener('keydown', this.handleKeyDown);
  }

  handleKeyDown = (event: KeyboardEvent) => {
    if (event.ctrlKey && event.key === 'x') {
      this.removeSelectedObject();
    }
  };

  removeSelectedObject = () => {
    const { selectedNodes } = this.state;
    if (selectedNodes.length === 0) {return;}

    const objectId = selectedNodes[0]; // Assuming only one object is selected

    const sharedModel = this.props.cpModel.jcadModel?.sharedModel;
    if (!sharedModel) {
      return;
    }

    const dependants = sharedModel.getDependants(objectId);

    let body: React.JSX.Element;
    if (dependants.length) {
      body = (
        <div>
          {'Removing this object will also result in removing:'}
          <ul>
            {dependants.map(dependant => (
              <li key={dependant}>{dependant}</li>
            ))}
          </ul>
        </div>
      );
    } else {
      body = <div>Are you sure?</div>;
    }

    showDialog({
      title: `Removing ${objectId}`,
      body,
      buttons: [Dialog.okButton(), Dialog.cancelButton()]
    }).then(({ button: { accept } }) => {
      if (accept) {
        const toRemove = dependants.concat([objectId]);
        sharedModel.removeObjects(toRemove);
      }
    });

    this.props.cpModel.jcadModel?.syncSelected({});
  };
```
And it works, I just copied the body of `onClick()` inside a function `removeSelectedObject`, of the remove object button in the `ObjectTree` component & invoking that on KeyboardEvent. Again this is a rough approach which doesn't follow DRY principle yet but I can work on standardizing the same if this is what is expected by the delete key binding.

trungleduc, 2024-07-24:
you might want to take a look at this PR https://github.com/QuantStack/jupytergis/pull/58. jupytergis and jupytercad share a lot of codebase

arjxn-py, 2024-07-24:
Thanks, this helps a lot.

PR #388 (2024-07-30): Delete Object Keyboard Shortcut
Should fix #46 
Adds a `Ctrl`+`X` keybinding to delete object.
closed 2024-07-30 by martinRenou