jupytercad/JupyterCAD #101: `@deathbeds/jupyterlab-rjsf` is not compatible with JupyterLab 3

trungleduc, 2023-01-18:
`@deathbeds/jupyterlab-rjsf` creates forms from schema files with a JupyterLab look and feel. It is compatible only with JupyterLab 3, we need to pin some dependencies to make it work in JLab4 but it is not a long-term solution. 
e need to update this library to make it compatible with both JLab 3 and 4.

bollwyvl, 2023-01-23:
yep, the code, as-is, isn't probably "core-grade," whatever that means, but i certainly wouldn't mind _not_ maintaining it.

that being said, it really does solve a huge swath of UI concerns, and could replace a ton of code. i of course dream of a day where one only need write a properly-described command, and get a a free UI.

both on the initial settings editor and the more recent metadata form, i encouraged the authors to look to this older implementation and make a generalized, lumino-friendly, well-behaved, lazy-loading, markdown-rendering tool that folk could reuse in sidebars, main areas, dialogs (ick), etc. with a predictable way to advertise new fields and widgets (ah, yes, _another_ kind of widget) to the registry. it's rather a big dependency with some nasty global state, so having multiple copies around is bad for business.

as i said on https://github.com/deathbeds/jupyterlab-starters/issues/97: i'm waiting to update to rjsf 5 to spend any more time on it, which should be out Real Soon Now :tm:, but this will be complicated by also wanting to get some things like https://github.com/jupyterlab/jupyterlab/issues/13746. but anyhow: i can't don't have a lot of time for downstreams of lab 4 just yet.

bollwyvl, 2023-01-31:
[rjsf 5 is out](https://github.com/rjsf-team/react-jsonschema-form/releases/tag/v5.0.0).

I think trying to get that upgraded, and create a useful `SchemaForm` for extension authors, should be prioritized for lab 4, and the deathbeds one can just be for 3.x.

trungleduc, 2023-01-31:
> We can probably fallback to using the same theme as the JupyterLab core settings editor for now, so that at least JupyterCad works for Lab 4. And long term solution would be to move `@deathbeds/jupyterlab-rjsf` to core?

👍 on dropping `@deathbeds/jupyterlab-rjsf`

martinRenou, 2025-04-29:
closing as outdated!

closed 2025-04-29 by martinRenou