jupytercad/JupyterCAD #132: Pressing "Enter" in a form does not validate the form

martinRenou, 2023-04-26:
When filling a form *e.g.* for creating a box, pressing "Enter" does not validate the form, it discards it. We should have "Esc" discard the form and "Enter" validate it.

strategos-nzr, 2023-06-03:
[image]

It is the same for 'submit' buttons as well. 

Also in the sketcher UI there is no way (even with escape) to re-select or de-select points or sketch tools.

Is it possible to import the FreeCAD gui and save work?

trungleduc, 2023-06-03:
IIRC you can double click to remove the points

strategos-nzr, 2023-06-03:
on my machine (windows) its right click, which works most of the time

Is there a way to enter into a completed sketch? I only see the option to make a new sketch.

SylvainCorlay, 2024-10-16:
This seems to be fixed on my end.

arjxn-py, 2024-10-17:
Yeah, fixed on my end too

closed 2024-10-17 by trungleduc