jupytercad/JupyterCAD #170: Error during development install

davidbrochart, 2023-06-01:
I tried to do a development install but got this error:
```console
$ pip install -e .
...
$ jupyter labextension develop . --overwrite
/home/david/micromamba/envs/jupytercad/lib/python3.11/site-packages/jupyterlab/debuglog.py:56: UserWarning: An error occurred.
  warnings.warn("An error occurred.")
/home/david/micromamba/envs/jupytercad/lib/python3.11/site-packages/jupyterlab/debuglog.py:57: UserWarning: ModuleNotFoundError: There is no labextension at .. Errors encountered: [TypeError("the 'package' argument is required to perform a relative import for '.'"), ModuleNotFoundError("No module named 'jupytercad.notebook.objects._schema'"), ModuleNotFoundError("No module named 'jupytercad.notebook.objects._schema'"), ModuleNotFoundError("No module named 'jupytercad.notebook.objects._schema'"), ModuleNotFoundError("No module named 'jupytercad.notebook.objects._schema'"), ModuleNotFoundError("No module named 'jupytercad.notebook.objects._schema'"), ModuleNotFoundError("No module named 'jupytercad.notebook.objects._schema'"), ModuleNotFoundError("No module named 'jupytercad.notebook.objects._schema'")]
  warnings.warn(msg[-1].strip())
/home/david/micromamba/envs/jupytercad/lib/python3.11/site-packages/jupyterlab/debuglog.py:58: UserWarning: See the log file for details: /tmp/jupyterlab-debug-1ynnmm4d.log
  warnings.warn(f"See the log file for details: {log_path!s}")
```

davidbrochart, 2023-06-01:
> it probably time for a coffee break when that happens

Happening now :coffee:

davidbrochart, 2023-06-01:
Failed with:
```console
$ jlpm build
lerna notice cli v6.6.2
lerna info versioning independent

    ✔  @jupytercad/jupytercad-opencascade:build (6m)

    ✖  @jupytercad/jupytercad-extension:build
       command not found: datamodel-codegen
       

 ——————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————

 >  Lerna (powered by Nx)   Ran target build for 3 projects (6m)
 
    ✔    1/2 succeeded [0 read from cache]
 
    ✖    1/2 targets failed, including the following:
         - @jupytercad/jupytercad-extension:build
```

davidbrochart, 2023-06-01:
Ah yes I seem them in [pyproject.toml](https://github.com/QuantStack/jupytercad/blob/65c8ba939c9b1f86bf3dbd18f5d2eeb3fb67597e/pyproject.toml#L3-L8).

davidbrochart, 2023-06-01:
Maybe the build could be more integrated with `hatch` so that it happens in the build environment where the build dependencies are installed.

martinRenou, 2023-06-01:
The @jupytercad/jupytercad-opencascade build should be cached

davidbrochart, 2023-06-01:
Yes.
But I'm still not able to open a notebook with `jupyter cad`. Here are my installation steps:
```
micromamba create -n jupytercad
micromamba activate jupytercad
micromamba install -c conda-forge python nodejs freecad "python<3.10"
pip install -e .
jlpm install
pip install hatchling hatch-nodejs-version jupyterlab datamodel-code-generator
jlpm build
jupyter labextension develop . --overwrite
jupyter server extension enable jupytercad
jupyter cad
```

martinRenou, 2023-06-02:
You cannot open a Notebook with JupyterCAD, the dedicated CAD app was built before the Python API so it made sense but not anymore.

Anyway you can open classic JupyterLab and open Notebook files there.