jupytercad/JupyterCAD #183: Missing dependencies

roaldarbol, 2023-06-04:
Really looking forward to using JupyterCAD, looks like an amazing tool!

TL;DR: I made a clean install with `poetry`, but failed a couple of times along the way due to missing dependencies, so thought it would be useful to document (maybe they should be added as additional dependencies?). After each step I run `jupyter cad` - I've added the error messages for each step below:

<details>
<summary>1. After adding <code>jupyter_cad</code>: <code>jupyterlab_server</code> missing</summary>

```python
Traceback (most recent call last):
  File "/usr/local/bin/jupyter-cad", line 5, in <module>
    from jupytercad.cadapp import main
  File "/usr/local/lib/python3.10/site-packages/jupytercad/__init__.py", line 2, in <module>
    from .cadapp import CadApp
  File "/usr/local/lib/python3.10/site-packages/jupytercad/cadapp/__init__.py", line 1, in <module>
    from .cadapp import CadApp, main  # noqa
  File "/usr/local/lib/python3.10/site-packages/jupytercad/cadapp/cadapp.py", line 10, in <module>
    from jupyterlab_server import LabServerApp
ModuleNotFoundError: No module named 'jupyterlab_server'
```
</details>

<details>
<summary>2. After adding <code>jupyterlab_server</code>: <code>jupyterlab</code> missing</summary>

```python
(jp-cad-py3.10) (base) ➜  jp-cad jupyter cad
[I 2023-06-04 12:22:15.530 ServerApp] Package jupytercad took 0.0000s to import
[I 2023-06-04 12:22:15.693 ServerApp] Package jupyter_collaboration took 0.1624s to import
[I 2023-06-04 12:22:15.694 ServerApp] Package jupyter_server_fileid took 0.0000s to import
[I 2023-06-04 12:22:15.741 ServerApp] Package jupyter_server_terminals took 0.0462s to import
[W 2023-06-04 12:22:15.743 ServerApp] jupyterlab | error adding extension (enabled: True): The module 'jupyterlab' could not be found (No module named 'jupyterlab'). Are you sure the extension is installed?
    Traceback (most recent call last):
      File "/Users/roaldarbol/Filen/git-projects/jp-cad/.venv/lib/python3.10/site-packages/jupyter_server/extension/manager.py", line 319, in add_extension
        extpkg = ExtensionPackage(name=extension_name, enabled=enabled)
      File "/Users/roaldarbol/Filen/git-projects/jp-cad/.venv/lib/python3.10/site-packages/jupyter_server/extension/manager.py", line 183, in __init__
        self._load_metadata()
      File "/Users/roaldarbol/Filen/git-projects/jp-cad/.venv/lib/python3.10/site-packages/jupyter_server/extension/manager.py", line 198, in _load_metadata
        raise ExtensionModuleNotFound(msg) from None
    jupyter_server.extension.utils.ExtensionModuleNotFound: The module 'jupyterlab' could not be found (No module named 'jupyterlab'). Are you sure the extension is installed?
[W 2023-06-04 12:22:15.744 ServerApp] nbclassic | error adding extension (enabled: True): The module 'nbclassic' could not be found (No module named 'nbclassic'). Are you sure the extension is installed?
    Traceback (most recent call last):
      File "/Users/roaldarbol/Filen/git-projects/jp-cad/.venv/lib/python3.10/site-packages/jupyter_server/extension/manager.py", line 319, in add_extension
        extpkg = ExtensionPackage(name=extension_name, enabled=enabled)
      File "/Users/roaldarbol/Filen/git-projects/jp-cad/.venv/lib/python3.10/site-packages/jupyter_server/extension/manager.py", line 183, in __init__
        self._load_metadata()
      File "/Users/roaldarbol/Filen/git-projects/jp-cad/.venv/lib/python3.10/site-packages/jupyter_server/extension/manager.py", line 198, in _load_metadata
        raise ExtensionModuleNotFound(msg) from None
    jupyter_server.extension.utils.ExtensionModuleNotFound: The module 'nbclassic' could not be found (No module named 'nbclassic'). Are you sure the extension is installed?
[W 2023-06-04 12:22:15.745 ServerApp] voila.server_extension | error adding extension (enabled: True): The module 'voila.server_extension' could not be found (No module named 'voila'). Are you sure the extension is installed?
    Traceback (most recent call last):
      File "/Users/roaldarbol/Filen/git-projects/jp-cad/.venv/lib/python3.10/site-packages/jupyter_server/extension/manager.py", line 319, in add_extension
        extpkg = ExtensionPackage(name=extension_name, enabled=enabled)
      File "/Users/roaldarbol/Filen/git-projects/jp-cad/.venv/lib/python3.10/site-packages/jupyter_server/extension/manager.py", line 183, in __init__
        self._load_metadata()
      File "/Users/roaldarbol/Filen/git-projects/jp-cad/.venv/lib/python3.10/site-packages/jupyter_server/extension/manager.py", line 198, in _load_metadata
        raise ExtensionModuleNotFound(msg) from None
    jupyter_server.extension.utils.ExtensionModuleNotFound: The module 'voila.server_extension' could not be found (No module named 'voila'). Are you sure the extension is installed?
[I 2023-06-04 12:22:15.756 ServerApp] jupyter_collaboration | extension was successfully linked.
[I 2023-06-04 12:22:15.785 ServerApp] jupyter_server_fileid | extension was successfully linked.
[I 2023-06-04 12:22:15.797 ServerApp] jupyter_server_terminals | extension was successfully linked.
[I 2023-06-04 12:22:15.808 ServerApp] jupytercad | extension was successfully linked.
[I 2023-06-04 12:22:15.913 ServerApp] jupyter_collaboration | extension was successfully loaded.
[I 2023-06-04 12:22:15.913 FileIdExtension] Configured File ID manager: ArbitraryFileIdManager
[I 2023-06-04 12:22:15.914 FileIdExtension] ArbitraryFileIdManager : Configured root dir: /Users/roaldarbol/Filen/git-projects/jp-cad
[I 2023-06-04 12:22:15.914 FileIdExtension] ArbitraryFileIdManager : Configured database path: /Users/roaldarbol/Library/Jupyter/file_id_manager.db
[I 2023-06-04 12:22:16.082 FileIdExtension] ArbitraryFileIdManager : Successfully connected to database file.
[I 2023-06-04 12:22:16.082 FileIdExtension] ArbitraryFileIdManager : Creating File ID tables and indices with journal_mode = DELETE
[I 2023-06-04 12:22:16.135 FileIdExtension] Attached event listeners.
[I 2023-06-04 12:22:16.135 ServerApp] jupyter_server_fileid | extension was successfully loaded.
[I 2023-06-04 12:22:16.137 ServerApp] jupyter_server_terminals | extension was successfully loaded.
[... 192 lines cut ...]
        http://localhost:8888/cad?token=9fa85750bf2b6c5a02b2226dd54a8828806cebff14f56a44
        http://127.0.0.1:8888/cad?token=9fa85750bf2b6c5a02b2226dd54a8828806cebff14f56a44
[I 2023-06-04 13:14:30.725 ServerApp] Skipped non-installed server(s): bash-language-server, dockerfile-language-server-nodejs, javascript-typescript-langserver, jedi-language-server, julia-language-server, pyright, python-language-server, python-lsp-server, r-languageserver, sql-language-server, texlab, typescript-language-server, unified-language-server, vscode-css-languageserver-bin, vscode-html-languageserver-bin, vscode-json-languageserver-bin, yaml-language-server
[I 2023-06-04 13:14:36.138 ServerApp] Creating new file in 
```

</details>

<details>
<summary>5.  <code>voila</code> can't install</summary>

```python
(jp-cad-py3.10) (base) ➜  jp-cad poetry add voila     
Using version ^0.4.0 for voila

Updating dependencies
Resolving dependencies... (0.5s)

Because no versions of voila match >0.4.0,<0.5.0
 and voila (0.4.0) depends on jupyter-server (>=1.18,<2.0.0), voila (>=0.4.0,<0.5.0) requires jupyter-server (>=1.18,<2.0.0).
And because jupyterlab (4.0.1) depends on jupyter-server (>=2.4.0,<3)
 and no versions of jupyterlab match >4.0.1,<5.0.0, voila (>=0.4.0,<0.5.0) is incompatible with jupyterlab (>=4.0.1,<5.0.0).
So, because jp-cad depends on both jupyterlab (^4.0.1) and voila (^0.4.0), version solving failed.
```
</details>

trungleduc, 2023-06-04:
Hi, can you try to install `JupyterCAD` with pip? All the dependencies should be installed automatically, I'm not a `peotry` user so I don't know if it reads the dependencies list of `JupyerCAD`

roaldarbol, 2023-06-04:
`poetry` doesn't do anything special compared to `pip`, it uses PyPi to install as default. I get the same output with direct `pip install jupytercad` (`jupyter lab` works, the error appears for `jupyter cad`). I get:
```python
(base) ➜  ~ jupyter cad
Traceback (most recent call last):
  File "/usr/local/bin/jupyter-cad", line 5, in <module>
    from jupytercad.cadapp import main
  File "/usr/local/lib/python3.10/site-packages/jupytercad/__init__.py", line 2, in <module>
    from .cadapp import CadApp
  File "/usr/local/lib/python3.10/site-packages/jupytercad/cadapp/__init__.py", line 1, in <module>
    from .cadapp import CadApp, main  # noqa
  File "/usr/local/lib/python3.10/site-packages/jupytercad/cadapp/cadapp.py", line 10, in <module>
    from jupyterlab_server import LabServerApp
ModuleNotFoundError: No module named 'jupyterlab_server'
```

From looking at the `pyproject.toml` file, `jupyterlab_server` is not there either, only `jupyter_server`. Where should the mentioned dependencies be mentioned?

martinRenou, 2023-06-05:
>  After adding nbclassic: IT WORKS - but voila missing

JupyterCad does not need Voila, does it?

closed 2023-06-05 by martinRenou

trungleduc, 2023-06-05:
Fixed in 0.2.3

roaldarbol, 2023-06-05:
Just did a clean re-install and works now on 0.2.3. 🥳

> JupyterCad does not need Voila, does it?

It seems to work without for sure, but still throws a warning - it seems to not impact performance, but I always get a bit nervous with any warnings thrown. ;-) And just FYI on 0.2.3 there are warnings for `nbclassic` and `voila`. The messages are the same as above.

martinRenou, 2023-06-05:
It seems you have server extensions (nbclassic and voila) for jupyter that are lying around but the related Python modules are not installed. Can you show the output of `jupyter server extension list`?

You may be able to remove those warnings by running `jupyter server extension disable {extension}` for extensions that are faulty.

roaldarbol, 2023-06-05:
Seems you're right about the competing installations... It seems to default back to the global installation (which came with Anaconda), even when I've got a `venv` installation. Here it first shows that it should use the `venv` installation, then I run `jupyter server extension list` which shows me a couple of different config dirs, and after that the chosen installation has changed. Really strange behaviour, but I reckon it's probably just a *me* issue - I'll try to figure it out. Thanks for your help though!

```python
(jp-cad-py3.10) (base) ➜  jp-cad which jupyter
/Users/roaldarbol/Filen/git-projects/jp-cad/.venv/bin/jupyter
(jp-cad-py3.10) (base) ➜  jp-cad jupyter server extension list
Config dir: /Users/roaldarbol/.jupyter

Config dir: /Users/roaldarbol/Filen/git-projects/jp-cad/.venv/etc/jupyter
    jupyter_lsp enabled
    - Validating jupyter_lsp...
Package jupyter_lsp took 0.1954s to import
A `_jupyter_server_extension_points` function was not found in jupyter_lsp. Instead, a `_jupyter_server_extension_paths` function was found and will be used for now. This function name will be deprecated in future releases of Jupyter Server.
      jupyter_lsp 2.2.0 OK
    jupyter_collaboration enabled
    - Validating jupyter_collaboration...
Package jupyter_collaboration took 6.2255s to import
      jupyter_collaboration 1.0.0 OK
    jupyter_server_fileid enabled
    - Validating jupyter_server_fileid...
Package jupyter_server_fileid took 0.0000s to import
      jupyter_server_fileid 0.9.0 OK
    jupyter_server_terminals enabled
    - Validating jupyter_server_terminals...
Package jupyter_server_terminals took 0.0902s to import
      jupyter_server_terminals 0.4.4 OK
    jupytercad enabled
    - Validating jupytercad...
Package jupytercad took 0.0000s to import
      jupytercad 0.2.3 OK
    jupyterlab enabled
    - Validating jupyterlab...
Package jupyterlab took 0.1014s to import
      jupyterlab 4.0.1 OK
    nbclassic enabled
    - Validating nbclassic...
Package nbclassic took 0.0038s to import
A `_jupyter_server_extension_points` function was not found in nbclassic. Instead, a `_jupyter_server_extension_paths` function was found and will be used for now. This function name will be deprecated in future releases of Jupyter Server.
      nbclassic 1.0.0 OK
    notebook_shim enabled
    - Validating notebook_shim...
Package notebook_shim took 0.0000s to import
A `_jupyter_server_extension_points` function was not found in notebook_shim. Instead, a `_jupyter_server_extension_paths` function was found and will be used for now. This function name will be deprecated in future releases of Jupyter Server.
      notebook_shim  OK

Config dir: /usr/local/etc/jupyter
    jupyter_collaboration enabled
    - Validating jupyter_collaboration...
Package jupyter_collaboration took 0.0000s to import
      jupyter_collaboration 1.0.0 OK
    jupyter_server_fileid enabled
    - Validating jupyter_server_fileid...
Package jupyter_server_fileid took 0.0000s to import
      jupyter_server_fileid 0.9.0 OK
    jupyter_server_terminals enabled
    - Validating jupyter_server_terminals...
Package jupyter_server_terminals took 0.0001s to import
      jupyter_server_terminals 0.4.4 OK
    jupytercad enabled
    - Validating jupytercad...
Package jupytercad took 0.0001s to import
      jupytercad 0.2.3 OK
    jupyterlab enabled
    - Validating jupyterlab...
Package jupyterlab took 0.0002s to import
      jupyterlab 4.0.1 OK
    nbclassic enabled
    - Validating nbclassic...
Package nbclassic took 0.0000s to import
A `_jupyter_server_extension_points` function was not found in nbclassic. Instead, a `_jupyter_server_extension_paths` function was found and will be used for now. This function name will be deprecated in future releases of Jupyter Server.
      nbclassic 1.0.0 OK
    voila.server_extension enabled
    - Validating voila.server_extension...
       X The module 'voila.server_extension' could not be found (No module named 'voila'). Are you sure the extension is installed?

(jp-cad-py3.10) (base) ➜  jp-cad which jupyter                
/usr/local/anaconda3/bin/jupyter
```

PR #184 (2023-06-05): Missing dependencies
Fix #183 By bringing back the jupyterlab dependency

We should figure out later why `jupyter cad` needs jupyterlab installed, and why `jupyter_core` and `jupyterlab_server` are not enough for it to work