jupytercad/JupyterCAD #192: Consider Supporting Related CodeCAD Projects (build123d and CadQuery)

jdegenstein, 2023-06-06:
Hello, I am involved with two open source projects that are also based on OCCT; build123d and CadQuery. These are both python-based "CodeCAD" projects for designing objects in code. 

Documentation for build123d:
https://build123d.readthedocs.io/en/latest/

Documentation for CadQuery
https://cadquery.readthedocs.io/en/latest/

Would you consider supporting these two projects in JupyterCAD? We have put a lot of effort into designing good user interfaces for CodeCAD, and we noticed that your project may benefit from supporting these two projects.

SylvainCorlay, 2023-06-07:
Quoting an email from @bernhard-42 (with his approval), who is the main author of [jupyter-cadquery](bernhard-42), one area in which the Python API of jupyer-cadquery is interesting is the enabling of algebraic expressions to operate on shapes.

> [algebra mode], a simple abstraction on top of open cascade (in fact on top of build123d’s direct API which simplifies open cascade) to code objects in an algebraic way: 
>    `(location1 * obj1 + location2 * obj2) - location3 * obj3)`, 
> which means fuse obj1 at location1 with object2 at location2 and then cut obj3 at location3 from the result, see https://build123d.readthedocs.io/en/latest/introductory_examples.html and https://build123d.readthedocs.io/en/latest/algebra_definition.html.

Jojain, 2023-06-07:
> Quoting an email from @bernhard-42 (with his approval), who is the main author of [jupyter-cadquery](bernhard-42), one area in which the Python API of jupyer-cadquery is interesting is the enabling [...]

The original work from @bernard-42 has been merged and improved, directly in `build123d` so support it comes with the algebra mode !

bernhard-42, 2023-06-08:
For context, adding the full statement that I sent to @SylvainCorlay 

> I looked at the current API and at a first glance it looks like a fluent API (similar to CadQuery). Build123d https://github.com/gumyr/build123d was started to get rid of the fluent API, since many users (especially non Python makers) struggled with it.
> 
> I contributed the algebra mode to it, a simple abstraction on top of open cascade (in fact on top of build123d’s direct API which simplifies open cascade) to code objects in an algebraic way: 
>     (location1 * obj1 + location2 * obj2) - location3 * obj3)”, 
> which means fuse obj1 at location1 with object2 at location2 and then cut obj3 at location3 from the result, see https://build123d.readthedocs.io/en/latest/introductory_examples.html and https://build123d.readthedocs.io/en/latest/algebra_definition.html.
> 
> Not saying that the builder or algebra mode in build123d are the correct ways of abstraction. However, reading through many discussions in the last few months around this topic, I believe that coming up with a simple and concise API to hide the complexity of e.g. open cascade is a very important part of an open source CAD application and its success.
> The advantage of JupyterCAD is that it is GUI based - however, when you fall back to code, it shouldn’t be a pain.
>

jgunstone, 2024-04-25:
it would also be great to support topologic which is also based on opencascade: 
https://github.com/wassimj/Topologic
https://github.com/wassimj/topologicpy

stefanv, 2024-11-10:
I'm curious if there were any further discussions along these lines? The algebra API for build123d is fantastic, and I'd love to see a powerful abstraction like that supported in JupyterCAD. Thanks, everyone, for all the hard work you've put into these tools!