jupytercad/JupyterCAD #196: Not displaying in JupyterLab 4.0.2

AlexGriffiths526, 2023-06-09:
I can't get anything to display in JupyterLab. I have taken a straightforward example from the API reference. See the screenshot below. The code runs and there is an output box like there should be something visible but is not. 

[image]

AlexGriffiths526, 2023-06-09:
Below are the logs - filtered to just include the errors/warnings. Please let me know if you need the full output. 

jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1  Plugin 'jupytercad:fcplugin' failed to activate.
(anonymous) @ jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1
jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1  TypeError: No provider for: @jupyter/collaboration-extension:ICollaborativeDrive.
    at H.resolveRequiredService (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1600442)
    at jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1599410
    at Array.map (<anonymous>)
    at H.activatePlugin (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1599397)
    at jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1600967
    at Array.map (<anonymous>)
    at H.start (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1600954)
    at K (812.8b0e8939a4ace07ad5c8.js?v=8b0e8939a4ace07ad5c8:1:8402)
(anonymous) @ jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1
jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1  Plugin 'jupytercad:jcadplugin' failed to activate.
(anonymous) @ jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1
jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1  TypeError: No provider for: @jupyter/collaboration-extension:ICollaborativeDrive.
    at H.resolveRequiredService (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1600442)
    at jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1599410
    at Array.map (<anonymous>)
    at H.activatePlugin (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1599397)
    at jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1600967
    at Array.map (<anonymous>)
    at H.start (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1600954)
    at K (812.8b0e8939a4ace07ad5c8.js?v=8b0e8939a4ace07ad5c8:1:8402)
(anonymous) @ jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1
jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1  Language pack 'English_United Kingdom' not valid!
fetch @ jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1
2jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1  Uncaught TypeError: Cannot read properties of null (reading 'type')
    at u._cellToolbarOverlapsContents (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:212598)
    at jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:212488
jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1  Connection lost, reconnecting in 0 seconds.
_reconnect @ jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1
jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1  TypeError: Cannot read properties of undefined (reading 'widget')
    at isReady (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1054584)
    at jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:873010
    at b (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:873104)
    at P.onKernelChanged (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1053741)
    at m (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1646609)
    at Object.l [as emit] (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1646285)
    at a.emit (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:1644122)
    at b.restartKernel (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:127006)
    at async y.restart (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:134704)
    at async execute (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:65676)
onKernelChanged @ jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1
3jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1  Uncaught (in promise) Error: CodeMirrorEditor already set.
    at ie.setEditor (jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:344082)
    at jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1:228104

AlexGriffiths526, 2023-06-09:
And the image in case this is easier to read. 
[image]

trungleduc, 2023-06-09:
Hi, could you check your `jupytercad` version?

AlexGriffiths526, 2023-06-09:
I am using Jupytercad 0.2.3

JupyterLab v4.0.2
C:\Users\AG\AppData\Local\anaconda3\envs\Jupyter4\share\jupyter\labextensions
        jupyterlab_pygments v0.2.2 enabled  X (python, jupyterlab_pygments)
        @jupyter/collaboration-extension v1.0.0 enabled ok (python, jupyter_collaboration)
        @jupytercad/jupytercad-extension v0.2.3 enabled ok (python, jupytercad)

   The following extensions are outdated:
        jupyterlab_pygments
 Consider running "jupyter labextension update --all" to check for updates.

Disabled extensions:
    @jupyterlab/filebrowser-extension:defaultFileBrowser

martinRenou, 2023-06-12:
I'm clueless here.

`TypeError: No provider for: @jupyter/collaboration-extension:ICollaborativeDrive.` would indicate that jupyter_collaboration is not properly installed, but it seems to be found by your jupyterlab installation

martinRenou, 2023-06-13:
Could you tell us which jupyter_collaboration version you have installed?

Also can you try installing jupytercad from conda-forge?

nthiery, 2023-06-13:
<details>
<summary installed packages>
# packages in environment at /opt/mambaforge/envs/jupytercad:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiofiles                  22.1.0             pyhd8ed1ab_0    conda-forge
aiosqlite                 0.19.0             pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
anyio                     3.7.0              pyhd8ed1ab_1    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
argon2-cffi               21.3.0             pyhd8ed1ab_0    conda-forge
argon2-cffi-bindings      21.2.0          py310h5764c6d_3    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
async-lru                 2.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
babel                     2.12.1             pyhd8ed1ab_1    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
beautifulsoup4            4.12.2             pyha770c72_0    conda-forge
bleach                    6.0.0              pyhd8ed1ab_0    conda-forge
boost-cpp                 1.74.0               h75c5d50_8    conda-forge
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.19.1               hd590300_0    conda-forge
ca-certificates           2023.5.7             hbcca054_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h255011f_3    conda-forge
charset-normalizer        3.1.0              pyhd8ed1ab_0    conda-forge
coin3d                    4.0.0             hc543bf6_1011    conda-forge
comm                      0.1.3              pyhd8ed1ab_0    conda-forge
curl                      8.1.2                h409715c_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
debugpy                   1.6.7           py310heca2aa9_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
entrypoints               0.4                pyhd8ed1ab_0    conda-forge
exceptiongroup            1.1.1              pyhd8ed1ab_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
ffmpeg                    4.4.2           gpl_h8dda1f0_112    conda-forge
fftw                      3.3.10          nompi_hc118613_108    conda-forge
flann                     1.9.1             he05ef13_1011    conda-forge
flit-core                 3.9.0              pyhd8ed1ab_0    conda-forge
fltk                      1.3.8                h83e168f_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freecad                   0.20.2          py310h8dc7af3_0    conda-forge
freeimage                 3.18.0              h530e30e_10    conda-forge
freetype                  2.12.1               hca18f0e_1    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gdk-pixbuf                2.42.10              h5eee18b_0  
gettext                   0.21.1               h27087fc_0    conda-forge
giflib                    5.2.1                h0b41bf4_3    conda-forge
gl2ps                     1.4.2                h0708190_0    conda-forge
glew                      2.1.0                h9c3ff4c_2    conda-forge
glib                      2.76.3               hfc55251_0    conda-forge
[... 236 lines cut ...]
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.8.4                h0b41bf4_0    conda-forge
xorg-libxau               1.0.11               hd590300_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
[... cut ...]
yjs-widgets               0.3.4              pyhd8ed1ab_0    conda-forge
ypy-websocket             0.8.4              pyhd8ed1ab_0    conda-forge
ypywidgets                0.4.1              pyhd8ed1ab_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.15.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.2                h3eb15da_6    conda-forge

</details>

martinRenou, 2023-06-13:
@nthiery would you be able to show us the very first errors you see in the JavaScript console?

AlexGriffiths526, 2023-06-13:
> I'm clueless here.
> 
> `TypeError: No provider for: @jupyter/collaboration-extension:ICollaborativeDrive.` would indicate that jupyter_collaboration is not properly installed, but it seems to be found by your jupyterlab installation.
> 
> Is this error `jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1 Plugin 'jupytercad:fcplugin' failed to activate. (anonymous) @ jlab_core.f68a597bc4700114bad4.js?v=f68a597bc4700114bad4:1` the first one you see in the JavaScript console? If there are other error messages higher up in the logs that may help debug.

Hiya, 
It was the first error yes. But I have it filtered to only show warnings and errors. 

I have opened up the JS console again today but strangely I have slightly different errors but still have the same issue. See log below. 
[image]