jupytercad/JupyterCAD #200: Have a UUID objects

martinRenou, 2023-06-13:
Renaming an existing object is problematic today if the object is used in a boolean operator.

We should have a UUID attached to each object that is **not** controlled by the user, and boolean operators would use this and not the object names as input.

martinRenou, 2024-12-02:
As discussed with @trungleduc and @arjxn-py:
- [ ] We might need to start introducing the schema version into the metadata field of jcad file
- [ ] We need to switch the underlying objects structure from a list to a dictionary
- [ ] we could support multiple schema versions at once

trungleduc, 2024-12-02:
The schema version check should be in `3.0.0`, so that in the future when users open a jcad file with a schema version greater than the supported one, they can get the error dialog.