jupytercad/JupyterCAD #207: Launch failure caused by TypeError __root__ instead of pydantic.RootModel

jonititan, 2023-07-12:
I've tried installing 0.3.1,0.3.2, & 0.3.3 into a clean new environment from conda but all fail to launch from the shell with the same error.
`
jupyter cad
Traceback (most recent call last):
  File "/home/jonititan/miniconda3/envs/cad/bin/jupyter-cad", line 6, in <module>
    from jupytercad.cadapp import main
  File "/home/jonititan/miniconda3/envs/cad/lib/python3.10/site-packages/jupytercad/__init__.py", line 4, in <module>
    from .notebook import *  # noqa
  File "/home/jonititan/miniconda3/envs/cad/lib/python3.10/site-packages/jupytercad/notebook/__init__.py", line 1, in <module>
    from .cad_document import CadDocument  # noqa
  File "/home/jonititan/miniconda3/envs/cad/lib/python3.10/site-packages/jupytercad/notebook/cad_document.py", line 15, in <module>
    from jupytercad.notebook.objects._schema.any import IAny
  File "/home/jonititan/miniconda3/envs/cad/lib/python3.10/site-packages/jupytercad/notebook/objects/__init__.py", line 8, in <module>
    from ._schema.jcad import Parts, ShapeMetadata  # noqa
  File "/home/jonititan/miniconda3/envs/cad/lib/python3.10/site-packages/jupytercad/notebook/objects/_schema/jcad.py", line 49, in <module>
    class JcadModel(BaseModel):
  File "/home/jonititan/miniconda3/envs/cad/lib/python3.10/site-packages/pydantic/_internal/_model_construction.py", line 95, in __new__
    private_attributes = inspect_namespace(
  File "/home/jonititan/miniconda3/envs/cad/lib/python3.10/site-packages/pydantic/_internal/_model_construction.py", line 279, in inspect_namespace
    raise TypeError("To define root models, use `pydantic.RootModel` rather than a field called '__root__'")
TypeError: To define root models, use `pydantic.RootModel` rather than a field called '__root__'
`

trungleduc, 2023-07-12:
Hi, can you try to downgrade `pydantic` to 1.10.11?

jonititan, 2023-07-13:
I  downgraded using 
```
conda install pydantic=1.10.11 
```
and it now works on both the linux and the windows machine.  

Thank you

closed 2023-07-13 by trungleduc

martinRenou, 2023-07-13:
I believe we should keep the issue open to track supporting the latest pydantic