jupytercad/JupyterCAD #209: Pydantic v2 breaks jupytercad

martinRenou, 2023-08-01:


gregnordin, 2023-08-15:
I just ran into this problem--when I execute `from jupytercad import CadDocument` in a jupyterlab notebook I get the following error message:

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[1], line 1
----> 1 from jupytercad import CadDocument

File ~/Documents/Projects/2023_Projects/jupytercad/.venv/lib/python3.11/site-packages/jupytercad/__init__.py:4
      1 from ._version import __version__  # noqa
      2 from .cadapp import CadApp
----> 4 from .notebook import *  # noqa
      7 def _jupyter_labextension_paths():
      8     return [{"src": "labextension", "dest": "@jupytercad/jupytercad-extension"}]

File ~/Documents/Projects/2023_Projects/jupytercad/.venv/lib/python3.11/site-packages/jupytercad/notebook/__init__.py:1
----> 1 from .cad_document import CadDocument  # noqa

File ~/Documents/Projects/2023_Projects/jupytercad/.venv/lib/python3.11/site-packages/jupytercad/notebook/cad_document.py:15
     12 from ypywidgets.ypywidgets import Widget
     14 from jupytercad.freecad.loader import fc
---> 15 from jupytercad.notebook.objects._schema.any import IAny
     16 from uuid import uuid4
     18 from .objects import (
     19     IBox,
     20     ICone,
   (...)
     29     ShapeMetadata,
     30 )

File ~/Documents/Projects/2023_Projects/jupytercad/.venv/lib/python3.11/site-packages/jupytercad/notebook/objects/__init__.py:8
      6 from ._schema.fuse import IFuse  # noqa
      7 from ._schema.intersection import IIntersection  # noqa
----> 8 from ._schema.jcad import Parts, ShapeMetadata  # noqa
      9 from ._schema.sphere import ISphere  # noqa
     10 from ._schema.torus import ITorus  # noqa

File ~/Documents/Projects/2023_Projects/jupytercad/.venv/lib/python3.11/site-packages/jupytercad/notebook/objects/_schema/jcad.py:49
     45     operators: Optional[List[Dict[str, Any]]] = None
     46     dependencies: Optional[List[str]] = None
---> 49 class JcadModel(BaseModel):
     50     __root__: List[JcadObject] = Field(..., title='IJCadModel')
     53 class JcadOptions(BaseModel):

File ~/Documents/Projects/2023_Projects/jupytercad/.venv/lib/python3.11/site-packages/pydantic/_internal/_model_construction.py:98, in ModelMetaclass.__new__(mcs, cls_name, bases, namespace, __pydantic_generic_metadata__, __pydantic_reset_parent_namespace__, **kwargs)
     96 config_wrapper = ConfigWrapper.for_model(bases, namespace, kwargs)
     97 namespace['model_config'] = config_wrapper.config_dict
---> 98 private_attributes = inspect_namespace(
     99     namespace, config_wrapper.ignored_types, class_vars, base_field_names
    100 )
    101 if private_attributes:
    102     if 'model_post_init' in namespace:
    103         # if there are private_attributes and a model_post_init function, we handle both

File ~/Documents/Projects/2023_Projects/jupytercad/.venv/lib/python3.11/site-packages/pydantic/_internal/_model_construction.py:291, in inspect_namespace(namespace, ignored_types, base_class_vars, base_class_fields)
    288 raw_annotations = namespace.get('__annotations__', {})
    290 if '__root__' in raw_annotations or '__root__' in namespace:
--> 291     raise TypeError("To define root models, use `pydantic.RootModel` rather than a field called '__root__'")
    293 ignored_names: set[str] = set()
    294 for var_name, value in list(namespace.items()):

TypeError: To define root models, use `pydantic.RootModel` rather than a field called '__root__'
```

martinRenou, 2023-08-21:
Please try to downgrade pydantic to v1 for now, an update for jupytercad will come soonish

closed 2023-11-09 by trungleduc