jupytercad/JupyterCAD #356: Extrusion produces wrong indexing -> visual artifacts

martinRenou, 2024-06-03:
Similar to https://github.com/jupytercad/jupytercad/issues/335 but for extrusion result

[image]

trungleduc, 2024-06-03:
hmm, could you test it with a previous version, maybe before https://github.com/jupytercad/jupytercad/pull/353

martinRenou, 2024-06-25:
> hmm, could you test it with a previous version, maybe before https://github.com/jupytercad/JupyterCAD/pull/353 ?

I also reproduce the same-ish issue before that PR

martinRenou, 2024-06-25:
It looks like it was working in `2.0.0a0` and stopped working in `2.0.0a1`

trungleduc, 2024-06-25:
wow, there are only 3 commits between a0 and a1, and two are meaningless. 

[image]

trungleduc, 2024-06-25:
actually, we are wrongly pinning the `jupytercad-*` package in the meta package. Installing `jupytercad` always gets the latest  `jupytercad-*` ones. This issue happens between a2 and a4, which contains the 3D refactoring pass. I will take a look at it

martinRenou, 2024-06-25:
Good catch!

trungleduc, 2024-06-25:
I managed to track it down into 2 issues:
- the front face and back face create additional meshes in the scene. https://github.com/jupytercad/JupyterCAD/blob/71e093ca9d979925396ee097d951a6d2ae438a44/packages/base/src/3dview/helpers.ts#L208 

[image]

It looks like these faces are related to the clipping panel, they are introduced in #333 

- if I remove it, I get a better one

[image]

it still has the face normal points to the inside, but I'm not sure if it's an issue or it's the behavior of OpenCascade, since if I extrude in the opposite direction, I get the face normal points to the outside.

[image]

martinRenou, 2024-06-26:
The clipping indeed relies on the normals, so if they are not correct, it results in visual artifacts.

> if I extrude in the opposite direction (LengthFwd=-10), I get the face normal points to the outside.

Interesting! Should we... reverse the normal depending on the sign of `LengthFwd`?

martinRenou, 2024-06-26:
What's weird though is that it seems to be 3D refactoring pass, you were not only changing the code structure right? You also fixed the normals for STL files https://github.com/jupytercad/JupyterCAD/pull/353, could this impact the extrusion?

trungleduc, 2024-06-26:
> What's weird though is that it seems to be coming from the 3D refactoring pass? You were not only changing the code structure right? You also fixed the normals for STL files #353, could this impact the extrusion?

I don't think it's related to the 3D pass, here is the result at https://github.com/jupytercad/JupyterCAD/commit/234f9bf6929261053a3887f3d1df57e8ed029c79, it works correctly

[image]

Here is the result one commit after, at https://github.com/jupytercad/JupyterCAD/commit/85c0dde2cc4d0e92f5d31dad7b027671762fff72

[image]

trungleduc, 2024-06-26:
> Interesting! Should we... reverse the normal depending on the sign of LengthFwd?

Or just using double-sided material?

trungleduc, 2024-06-26:
It looks like the additional front and back meshes are made for the equivalent 3D solid tube since there are meshes for the top and bottom circles.

[image]

martinRenou, 2024-06-26:
I'll have a look

trungleduc, 2024-06-26:
> I'll have a look. It seems the visibility toggle does not hide all the meshes created for a specific object (there are multiple ones to achieve the clipping).

 but we should not have the meshes at the top and bottom of the extruded shape ?

martinRenou, 2024-06-26:
> but we should not have the meshes at the top and bottom of the extruded shape ?

I'm not sure what you mean by that?

> I'm not sure what is the definition of correct normals

Always pointing outside?

Arf this is annoying... Clipping is what makes it difficult here because it needs proper normals, without this I would agree using double-sided materials and call it a day.

martinRenou, 2024-06-26:
As discussed offline, this issue comes from the clipping logic.

Clipping involves some logic to fill the gap left by the cut, this only makes sense for solid objects. However extrusion produces a non-solid object by default and that logic gets broken in that case, resulting in visual artifacts.

We should disable that logic for non-solid objects.