jupytercad/JupyterCAD #384: Backend undo manager

davidbrochart, 2024-07-23:
In jupyter-ydoc v3 there will be an [undo manager](https://github.com/jupyter-server/jupyter_ydoc/blob/54642267dc309be03c754a626aa75e0ac6f1c964/jupyter_ydoc/ybasedoc.py#L36) that will allow to undo/redo operations on a shared model in the backend in a document-agnostic way. Every document inheriting from `YBaseDoc` will be responsible for adding to the undo manager's scope the shared types that it wishes to be included. For instance, for a `YNotebook`, the [cells are included](https://github.com/jupyter-server/jupyter_ydoc/blob/54642267dc309be03c754a626aa75e0ac6f1c964/jupyter_ydoc/ynotebook.py#L59) in the undo manager's scope.
For a [YJCad](https://github.com/jupytercad/JupyterCAD/blob/d526259792a8cba1ba1e2d39194544fd8a1abfd3/python/jupytercad_core/jupytercad_core/jcad_ydoc.py#L9) document, it looks like `self._yobjects` should be included?

martinRenou, 2024-07-23:
Can we still register the undo manager front-end side for it to work in jupyterlite?

davidbrochart, 2024-07-23:
> Can we still register the undo manager front-end side for it to work in jupyterlite?

Yes, they are unrelated.

> For a [YJCad](https://github.com/jupytercad/JupyterCAD/blob/d526259792a8cba1ba1e2d39194544fd8a1abfd3/python/jupytercad_core/jupytercad_core/jcad_ydoc.py#L9) document, it looks like `self._yobjects` should be included?

I'm asking because @meriem-benismail could undo but not redo, when just adding `self._yobjects` to the scope. Is there something else that should be included?

trungleduc, 2024-07-23:
> I'm asking because @Meriem-BenIsmail could undo but not redo, when just adding self._yobjects to the scope. Is there something else that should be included?

Nope, the rendering logic is triggered on `__yobjects ` change. Did she get the new `__yobjects` in the frontend?

arjxn-py, 2024-07-23:
> could undo but not redo

Hi, I can confirm as I encountered the same scenario in [383#comment](https://github.com/jupytercad/JupyterCAD/pull/382#discussion_r1685430812)

Meriem-BenIsmail, 2024-07-23:
Hi, when expanding the scope of the undo manager for the [YJCad](https://github.com/jupytercad/JupyterCAD/blob/d526259792a8cba1ba1e2d39194544fd8a1abfd3/python/jupytercad_core/jupytercad_core/jcad_ydoc.py#L9) example like this 
` self.undo_manager.expand_scope(self._yobjects)`, the undo works but the same can't be said for the redo.
It seems the redo stack of the undo manager is always empty.

davidbrochart, 2024-07-23:
One thing that could explain this behavior,  is that JupyterCAD reacts to an undo operation by doing another operation on `self._yobjects`, which would clear the redo stack. Could that be the case?

trungleduc, 2024-07-23:
indeed, JupterCAD updates object metadata after rendering. 

https://github.com/jupytercad/JupyterCAD/blob/09f22d858c1cb968a6d8f9674d89a1d08408befa/packages/base/src/3dview/mainviewmodel.ts#L109

Meriem-BenIsmail, 2024-07-23:
The redo stack behavior isn't the only issue here. Whenever the objects are updated, this [method](https://github.com/jupytercad/JupyterCAD/blob/main/packages/base/src/3dview/mainviewmodel.ts#L243) should be invoked. It gets called during the undo operation, but not during the redo.

davidbrochart, 2024-07-23:
> indeed, JupterCAD updates object metadata after rendering.

But `self._ymetadata` is a separate shared type, so it is fine. It wouldn't be if `self._yobjects` were changed.

Meriem-BenIsmail, 2024-07-23:
It turns out the issue was indeed with `this._savemeta(result)` . When I commented it out, the redo worked properly.

trungleduc, 2024-07-23:
I'm wondering if the undo manager should support this use case?

trungleduc, 2024-07-23:
Here is how an object looks like. It's a standard `YMap` and the metadata is not defined as a shared type. 

https://github.com/jupytercad/JupyterCAD/blob/09f22d858c1cb968a6d8f9674d89a1d08408befa/packages/schema/src/schema/jcad.json#L83

```
    "jcadObject": {
      "title": "IJCadObject",
      "type": "object",
      "additionalProperties": false,
      "required": ["name", "visible"],
      "properties": {
        "name": {
          "type": "string"
        },
        "visible": {
          "type": "boolean"
        },
        "shape": {
          "$ref": "#/definitions/parts"
        },
        "parameters": {
          "type": "object"
        },
        "shapeMetadata": {
          "$ref": "#/definitions/shapeMetadata"
        },
        "operators": {
          "type": "array",
          "items": {
            "type": "object"
          }
        },
        "dependencies": {
          "type": "array",
          "items": {
            "type": "string"
          }
        }
      }
    },
```

davidbrochart, 2024-07-23:
Thanks.
Is `shapeMetadata` automatically computed from the other values? Where is the line between what can be changed e.g. by the user and what reacts to these changes?

trungleduc, 2024-07-23:
> Is shapeMetadata automatically computed from the other values?

Yes, the geometry properties are computed by Opencascade after rendering. Some other keys are defined at shape creation time. None are modifiable by users and there are no reactivities on these values.

davidbrochart, 2024-07-23:
Well I don't know this project enough, but if a user action on an object leads to a change of the same object, then it cannot work with an undo manager (at least it's not possible to redo after an undo).

trungleduc, 2024-07-23:
can we detect the origin of the modification? if it is from the undo manager we can skip the metadata update since it's correct anyway.

davidbrochart, 2024-07-23:
Good point. There is an [issue](https://github.com/jupyter-server/pycrdt/issues/120) for that, I can look into it.
But I don't think it will solve @arjxn-py's issue since his undo manager lives in the frontend and origins are not propagated.
I'm wondering why metadata is stored in the shared model if it can automatically be computed?

trungleduc, 2024-07-23:
>  it can automatically be computed

JupyterCAD needs OpenCascade in the frontend to compute these geometry properties, but we also want to have access to these properties from a kernel

davidbrochart, 2024-07-23:
In any case, if all the data were stored under a shared type, separate from the metadata, we could add that shared type to the undo manager's scope. Do you think it's possible?

trungleduc, 2024-07-23:
yes, I thought about this solution but I feel more comfortable with the update origin approach since it's more natural to have the object metadata in the object itself. But we can use this solution as the last resort

davidbrochart, 2024-07-23:
The metadata would still be in the object itself, but all the other (undoable) data would be under e.g. a `YMap`, so that we can pass it to an undo manager. Right now we can only pass the whole object itself, or even worse the list of objects.

davidbrochart, 2024-07-23:
> I feel more comfortable with the update origin approach

The problem with that is that origins don't propagate through the network, so if the metadata update happens in the frontend, it won't solve @Meriem-BenIsmail's issue, and if the metadata update happens in the backend, it won't solve @arjxn-py's issue :smile:

trungleduc, 2024-07-23:
something like this could work ?

```
_yobjects = YList[YMap[ data: YMap, metadata: YMap ]]
```
If it could, it's possible to update the shared model schema but I'm not sure about the amount of work.

davidbrochart, 2024-07-23:
Yes I think it would work (`metadata` doesn't need to be a `YMap`, only `data`).

davidbrochart, 2024-07-23:
I can work on origins in `pycrdt` in parallel, but I also don't know about the amount of work.

trungleduc, 2024-07-23:
> I can work on origins in `pycrdt` in parallel, but I also don't know about the amount of work.

if it's the same amount of work, yours would have a much greater impact 😺

davidbrochart, 2024-07-23:
As I said, it wouldn't work in all cases :mouse:

trungleduc, 2024-07-29:
@Meriem-BenIsmail could you test your undo manager with this #385 to see if the redo works?

PR #385 (2024-07-30): Improve shape metadata update logic
Fixes #384

https://github.com/user-attachments/assets/a510b967-50b5-49c2-86f0-02fe36a7b08b

## Code change

- Shape metadata is updated at the same time as the shape parameters
- Add tests for Undo and Redo function
closed 2024-07-30 by martinRenou