jupytercad/JupyterCAD #391: [ENH] Implement Cut/Copy/Paste Logic for Objects

arjxn-py, 2024-07-30:
Currently, the extension lacks cut, copy, and paste operations for objects, which are nice to have for enhancing UX through standard keyboard shortcuts. This feature request aims to implement these functionalities along with their respective key bindings.

**Cut (Ctrl/Cmd + X):**

- Remove the selected object from the current view and store it in the clipboard.
- This operation should be undoable.
- Add the keybinding `Accel X`.

**Copy (Ctrl/Cmd + C):**

- Copy the selected object to the clipboard without removing it from the current view.
- This operation should be undoable.
- Add the keybinding `Accel C`.

**Paste (Ctrl/Cmd + V):**

- Insert the object from the clipboard into the current view.
- This operation should place the object at the mouse cursor location or in a default position if no cursor location is applicable.
- This operation should be undoable.
- Add the keybinding `Accel V`.

martinRenou, 2024-11-28:
While playing more with JupyterCAD, I realized how badly this is missing :D

martinRenou, 2024-11-28:
I'm adding it to the 3.0.0 milestone, but we can always move it to a 3.1.0 milestone if we don't want to delay 3.0.0 with this

arjxn-py, 2024-12-02:
I'm giving it a try

martinRenou, 2024-12-03:
Done in #624 thanks!

closed 2024-12-03 by martinRenou