jupytercad/JupyterCAD #394: [ENH] Object Color Customization in JupyterCAD

arjxn-py, 2024-08-10:
Currently, JupyterCAD models have a single default color, which is gray. It would be highly beneficial to introduce a feature allowing users to change the color of individual objects within a model. This could be implemented using hex codes and a color picker, giving users the flexibility to customize the appearance of their models to suit their needs.

Proposed Solution:

- Introduce a color picker UI element in the right panel maybe.
- Allow users to input hex color codes in an input field in object properties form.
- Have an option to apply these colors to individual objects within the model.

arjxn-py, 2024-08-12:
I want to ask where are we setting the color of the objects. I ask because I've already looked into `mainview` but couldn't exactly find.

martinRenou, 2024-08-19:
Thanks for the suggestion! 

We have color support when loading FCSTD files (through the `guidata` entry in the model), but we don't allow editing those colors. Also we indeed don't provide this feature for the jcad format.

I believe we should be able to add a "Color" property in the objects schemas, similar to all objects having a "Placement": https://github.com/jupytercad/JupyterCAD/blob/main/packages/schema/src/schema/box.json#L23
By doing so, a color picker should be visible in the object properties form automatically (at least a text input box, but we could make it a color picker by tweaking rjsf similar to what we've done in JupyterGIS).

We would then need the `mainview` to support reading this color and applying it to the 3D mesh material.

I'm thinking we may ditch the guidata thingy for FCSTD files and set the color through that new "Color" property we would be introducing.

arjxn-py, 2024-08-19:
> I believe we should be able to add a "Color" property in the objects schemas, similar to all objects having a "Placement":

Thanks for the clarification, this sounds like a good start. I should give it a try & discuss more retrospectively.

martinRenou, 2024-09-11:
Closed as resolved, thanks @arjxn-py !

closed 2024-09-11 by martinRenou