jupytercad/JupyterCAD #396: RFD: Proposal to Add Keybindings for Additional JupyterCAD Commands

arjxn-py, 2024-08-15:
Currently, JupyterCAD supports keybindings for a few essential commands such as undo, redo, and removeObject. However, there are several other commands in JupyterCAD that would benefit from having dedicated keybindings to enhance UX.

Proposed Keybindings(Draft): 

- **New Sketch:** `Ctrl + Shift + N`
- **New Box:** `Ctrl + Shift + B`
- **New Cylinder:** `Ctrl + Shift + C`
- **New Sphere:** `Ctrl + Shift + S`
- **New Cone:** `Ctrl + Shift + E`
- **New Torus:** `Ctrl + Shift + T`
- **Cut:** `Ctrl + X`
- **Extrusion:** `Ctrl + Shift + E`
- **Union:** `Ctrl + U`
- **Intersection:** `Ctrl + I`
- **Chamfer:** `Ctrl + Shift + F`
- **Fillet:** `Ctrl + F`
- **Update Axes:** `Ctrl + Shift + A`
- **Update Exploded View:** `Ctrl + Shift + V`
- **Update Camera Settings:** `Ctrl + Shift + K`
- **Update Clip View:** `Ctrl + Shift + L`
- **Export Jcad:** `Ctrl + Shift + X`

arjxn-py, 2024-08-15:
> * **New Box:** `Ctrl + Shift + B`
> * **New Cylinder:** `Ctrl + Shift + C`
> * **New Sphere:** `Ctrl + Shift + S`
> * **New Cone:** `Ctrl + Shift + E`
> * **New Torus:** `Ctrl + Shift + T`

I think these may also be fine to skip

trungleduc, 2024-08-19:
Thanks for the suggestion. I think we should keep it consistent by prefixing all shortcuts with Ctrl + Shift and should avoid collision with JupyterLab shortcuts 

[image]

trungleduc, 2024-08-19:
> > * **New Box:** `Ctrl + Shift + B`
> > * **New Cylinder:** `Ctrl + Shift + C`
> > * **New Sphere:** `Ctrl + Shift + S`
> > * **New Cone:** `Ctrl + Shift + E`
> > * **New Torus:** `Ctrl + Shift + T`
> 
> I think these may also be fine to skip

I have no issues keeping these shortcuts

PR #451 (2024-10-11): Add all proposed Keyboard shortcuts
Should close #396 
Should also close #458 

Trust me! I'm pressing the keybinding while being on the the notebook 😇

https://github.com/user-attachments/assets/cb919298-8d7e-4ac6-b394-c70be1b63f45
closed 2024-10-11 by SylvainCorlay