jupytercad/JupyterCAD #399: BUG: Error when Running JupyterCAD through `jupyter cad` Command

arjxn-py, 2024-09-05:
When running JupyterCAD through the `jupyter cad` command in the terminal, the following errors are encountered in the browser logs, and the JupyterCAD editor fails to open:

```
Error: Shared module @jupyterlab/codeeditor doesn't exist in shared scope default
```
```
Error: Shared module @jupyterlab/completer doesn't exist in shared scope default
```
However, this issue does not occur when JupyterCAD is accessed through the `jupyter lab` command. In that case, the editor opens successfully without any errors.

martinRenou, 2024-09-06:
This may just be about adding `@jupyterlab/codeeditor` and `@jupyterlab/completer` as dependencies for https://github.com/jupytercad/JupyterCAD/blob/main/python/jupytercad_app/package.json

trungleduc, 2024-09-06:
and also here https://github.com/jupytercad/JupyterCAD/blob/main/python/jupytercad_app/src/sharedscope.ts

arjxn-py, 2024-09-06:
Thanks both!
I'll be on it later today.

PR #402 (2024-09-09): Fix python console in the standalone app
- Add missing tokens to make python console work in the standalone app
- Closes #399
- Needs https://github.com/QuantStack/yjs-widgets/pull/14
closed 2024-09-09 by trungleduc