jupytercad/JupyterCAD #426: Deleting operators does not restore visibility of components

SylvainCorlay, 2024-09-16:
When deleting e.g. a Union operator, we may want either make the components of the union become visible again.

SylvainCorlay, 2024-09-16:
I am tagging this as a "question" because I am not sure what the desired behavior should be. Maybe we should look at what other editors do.

arjxn-py, 2024-09-16:
> I am tagging this as a "question" because I am not sure what the desired behavior should be. Maybe we should look at what other editors do.

Thanks @SylvainCorlay, I at the moment do not have a strong opinion on this but I'll be happy to see how FreeCAD handles it and report back :)

martinRenou, 2024-09-17:
> We may want to replicate the same in JCAD too

Sounds good!

SylvainCorlay, 2024-10-10:
> We may want to replicate the same in JCAD too

I agree that replicating the FreeCAD behavior is probably what we want here.

martinRenou, 2024-10-10:
We now have a `dependencies` property for all objects: https://github.com/jupytercad/JupyterCAD/blob/main/packages/schema/src/schema/jcad.json#L110

This could be used by setting `visible` back to True for all the `dependencies` objects as soon as the operator is removed.

closed 2024-10-24 by martinRenou