jupytercad/JupyterCAD #427: "Color: should be a string" error in operator form

SylvainCorlay, 2024-09-16:
When creating e.g. a union operator, I see the warning "Color: should be a string" in the form for the parameters of the operator.

I suppose there could be some default value assigned to the field so that it does not trigger an error.

[image]

arjxn-py, 2024-09-16:
This was also reported in https://github.com/jupytercad/JupyterCAD/pull/413#issuecomment-2348806223 but we couldn't reproduce it further 🫤 

I shall try looking into this more, should be an easy fix as far as I anticipate once reproduced. Thanks for opening this.

> I suppose there could be some default value assigned to the field so that it does not trigger an error.

Yes, it basically inherits the color of the base object which is being operated on.

arjxn-py, 2024-09-16:
@SylvainCorlay, can you maybe confirm how you got this error: 
- You must have created two objects
- Then straight up clicked on any operator button. (Did you select any of the object before clicking on the operator button?)

arjxn-py, 2024-09-16:
> Did you select any of the object before clicking on the operator button?

I ask as I could produce this another error when I haven't selected any object even once (related but not exactly this): 

```
Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'value')
```

I think this one error is because in case of never selected `model.localState?.selected` should be undefined.

Having said that this is a different error, I'd once again request you to try reproducing `Color: should be a string` error and maybe give some inference or observation that can help further. Apologies & Thanks.

SylvainCorlay, 2024-09-18:
> Having said that this is a different error, I'd once again request you to try reproducing Color: should be a string error and maybe give some inference or observation that can help further. Apologies & Thanks.

This seems to occur when opening an old jcad file dating from before colors were part of the model. The color is missing in the first object of the union.

We may decide to ignore this issue at this point.

arjxn-py, 2024-09-18:
> We may decide to ignore this issue at this point.

Sounds good to me, As it's not affecting new files.

closed 2024-09-18 by trungleduc