jupytercad/JupyterCAD #443: Improve UX of the exploded view toolbar button

martinRenou, 2024-10-02:
Similar to making the clip-plane and wireframe buttons toggle-able https://github.com/jupytercad/JupyterCAD/pull/441 we should look into making the exploded view button toggle-able too.

I'm thinking that once the exploded view is toggled, we could see the exploded view factor input widget on the 3D view itself as per the image below (sorry for the very bad editing 😅).

As of today the widget is an <input> tag, but I'm also thinking a slider could be nice.

[image]

arjxn-py, 2024-10-03:
Same could be done for AxesHelper too.

martinRenou, 2024-10-16:
As suggested by @SylvainCorlay 

> I propose to make the exploded view toolbar button a toggle.
> 
> The exploded view factor number input could be set either in a separate dropdown in the toolbar (at first), or in a text input overlayed on the view.
> 
> On a separate note, maybe there should be a visual indication in the view that the exploded view is not the normal position of objects - and how to leave this view, in case someone does not know how to set it back to normal.

martinRenou, 2024-10-16:
> indication in the view that the exploded view is not the normal position of objects

There is actually!

arjxn-py, 2024-12-03:
https://jupyterlab-contrib.github.io/jupyter-ui-toolkit/?path=/docs/components-slider--documentation - We're going to use this slider to set exploded factor

martinRenou, 2025-01-14:
https://github.com/geojupyter/jupytergis/pull/325 as a proof the slider from the ui-toolkit is quite nice :D

martinRenou, 2025-04-28:
Closed as done in #722 ! Thanks @HaudinFlorence

closed 2025-04-28 by martinRenou