jupytercad/JupyterCAD #458: Redo keyboard short

SylvainCorlay, 2024-10-08:
The redo keyboard shortcut should probably be consistent with that of the notebook editor and with the shortcut indicated in the "Edit" topbar menu, that is ⇧ ⌘ Z on OSX.

[image]

martinRenou, 2024-10-08:
For the record, this is where the keybinding is defined https://github.com/jupytercad/JupyterCAD/blob/main/packages/base/src/keybindings.json#L9

arjxn-py, 2024-10-08:
I shall look into how can i have different keybinding for OSX and Windows, Do you know already by any chance?

SylvainCorlay, 2024-10-08:
I think that we should be as close as possible to the notebook user experience for this.

arjxn-py, 2024-10-08:
> I think that we should be as close as possible to the notebook user experience for this.

I agree, and will be On my way! to see how they're handling it upstream. And I believe it shall not be an issue anymore to have the same keybinding for `redo` both in normal notebook or jcad file as conflict issue shall resolve in #451

arjxn-py, 2024-10-09:
I couldn't find `Ctrl Y` or `Accel Y` for redo in jupyterlab's code, should I just update this to `Accel Shift Z` in that case?

martinRenou, 2024-10-09:
Indeed JupyterLab is never using `Ctrl Y` for redo, let's stay aligned with JupyterLab

PR #451 (2024-10-11): Add all proposed Keyboard shortcuts
Should close #396 
Should also close #458 

Trust me! I'm pressing the keybinding while being on the the notebook 😇

https://github.com/user-attachments/assets/cb919298-8d7e-4ac6-b394-c70be1b63f45
closed 2024-10-11 by SylvainCorlay