jupytercad/JupyterCAD #462: Logo design

martinRenou, 2024-10-08:
Opening this for discussion.

@IsabelParedes made some nice logo designs 🚀 

@trungleduc @SylvainCorlay @arjxn-py please tell us your thoughts about them!

I have a personal preference for the green/yellow one.

Here there are:

# Blue

[image]

# Red

[image]

# Green/Yellow

[image]

# Orange

[image]

trungleduc, 2024-10-08:
Blue one for me

IsabelParedes, 2024-10-09:
Here are some python color logos as @arjxn-py suggested :)

[image]
[image]
[image]
[image]

arjxn-py, 2024-10-09:
Thanks @IsabelParedes, these are super awesome 🚀 
I really like first and third starting from left. 

Here's how they look, in their respective order:

[image]

[image]

[image]

[image]

SylvainCorlay, 2024-10-09:
This is really cool!

SylvainCorlay, 2024-10-10:
We probably still want to use the css styling of the svg in the file browser so that the logo becomes white when the corresponding line is highlighted.

trungleduc, 2024-10-10:
> Here are some python color logos as [@arjxn-py](https://github.com/arjxn-py) suggested :)
> 
> <img alt="" width="20%" src="https://private-user-images.githubusercontent.com/23283243/374939847-a86f448d-72a3-4509-8e6f-d3f8a660743a.svg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg1NDcyOTQsIm5iZiI6MTcyODU0Njk5NCwicGF0aCI6Ii8yMzI4MzI0My8zNzQ5Mzk4NDctYTg2ZjQ0OGQtNzJhMy00NTA5LThlNmYtZDNmOGE2NjA3NDNhLnN2Zz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEwVDA3NTYzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFlOWJmMjA1YWRjOWUxNGZjNjMyYmJhYjlkMGMzM2ZkMzVjMDJmNmQ5NjI1NmFiMDhlNjAxYzc3MzJiNzJjYTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.FbnbHPEXYEhbhX0bQN0uM0vFkTCdSQ9zxrLvmU7UQXo"> <img alt="" width="20%" src="https://private-user-images.githubusercontent.com/23283243/374939754-46d373be-1cc8-44c2-975e-a45eaa53ec3f.svg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg1NDcyOTQsIm5iZiI6MTcyODU0Njk5NCwicGF0aCI6Ii8yMzI4MzI0My8zNzQ5Mzk3NTQtNDZkMzczYmUtMWNjOC00NGMyLTk3NWUtYTQ1ZWFhNTNlYzNmLnN2Zz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEwVDA3NTYzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI1Mjg0MWI2OTA2NjM4MjIzN2QyNjY0Yjc0OGEwM2IwODViMjcyZTFiMGI3MGY4YzllOWMwN2YzNjBhZmY2OWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.1niSA5VhWAEsoEFq1PRn2aFOuBBOmIKJb6yhvBcJCtA"> <img alt="" width="20%" src="https://private-user-images.githubusercontent.com/23283243/374939893-61c5c381-3245-420c-ba2b-2e9e83624365.svg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg1NDcyOTQsIm5iZiI6MTcyODU0Njk5NCwicGF0aCI6Ii8yMzI4MzI0My8zNzQ5Mzk4OTMtNjFjNWMzODEtMzI0NS00MjBjLWJhMmItMmU5ZTgzNjI0MzY1LnN2Zz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEwVDA3NTYzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ0YzdjYzQwMmNjNjQyNzk0NGFjNGNjMTJkNWQxYjQ5NTRkNjY1MDJlYzliNTIxMDI3MmJlM2M3NjcxMDUwNzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.1_6Y2BrJPF1SVB65zRfX8xUneuEWtmQTb-VmGE3SoiA"> <img alt="" width="20%" src="https://private-user-images.githubusercontent.com/23283243/374939920-259433b1-7a7e-43b4-b387-ff0279440d35.svg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg1NDcyOTQsIm5iZiI6MTcyODU0Njk5NCwicGF0aCI6Ii8yMzI4MzI0My8zNzQ5Mzk5MjAtMjU5NDMzYjEtN2E3ZS00M2I0LWIzODctZmYwMjc5NDQwZDM1LnN2Zz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEwVDA3NTYzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU0MWRhYzYzZTVmMmU2ZjZmZmYwY2JmZmY0YzNiZDZjNjRlZWZiNGQ1OGZhYWZkZGRjZDg3Zjg5ZmQ4NjNjNDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.GrSOAYRwJKif_jiIJv0z5d3pPpc7gLlTk3WHmz-ga2U">

it looks great!

trungleduc, 2024-10-10:
> Thanks [@IsabelParedes](https://github.com/IsabelParedes), these are super awesome 🚀 I really like first and third starting from left. But if i have to choose one my preference would be first as it's more identifiable when small file icon due to the color difference between gears.

Does it work well in the dark theme?

arjxn-py, 2024-10-10:
Yes @trungleduc, it looks good to me. What do you think?
[image]
[image]
[image]
[image]

martinRenou, 2024-10-17:
Shall we go with the Python color palette then? I'm 👍🏽 for it, with a slight preference for the first screenshot from Arjun's comment right above.

arjxn-py, 2024-10-20:
> Shall we go with the Python color palette then? I'm 👍🏽 for it, with a slight preference for the first screenshot from Arjun's comment right above.

Sounds good to me :)

PR #520 (2024-10-22): Update main logo
Closes #462 
Fixes https://github.com/jupytercad/JupyterCAD/issues/484

[image]
closed 2024-10-22 by martinRenou