jupytercad/JupyterCAD #468: Dynamic Selection Colors Based on Object Color

arjxn-py, 2024-10-11:
Currently, the selection color is static and defaults to blue. While this works, it may not always provide the best visual clarity, especially when working with a variety of object colors. The static blue selection color can sometimes clash with the colors of certain objects, making it difficult to distinguish selections and reducing overall visual appeal.

We suggest making the selection color dynamic, where the selection highlight color is based on the object's original color. 
A slight alteration in brightness or saturation could be used to highlight the selected object without drastically changing its appearance.

martinRenou, 2024-10-14:
Thanks for starting an issue for this! Indeed we should improve the visual indication about selection.

I kind of like the threejs editor approach of displaying the bounding box of the selected mesh (see video). Also the fact that position controls are showing up upon selection helps a lot (see the work started in https://github.com/jupytercad/JupyterCAD/pull/151). If we take this approach, I'm not sure dynamic coloring is really needed.

https://github.com/user-attachments/assets/7978fe76-693e-4afb-bbd6-b7f314122115

arjxn-py, 2024-10-14:
Thanks @martinRenou, this also makes much sense and I'll try to put together a draft for this bounding box thing and update.

PR #473 (2024-10-16): Bounding Box for Selection

https://github.com/user-attachments/assets/887b59da-aaf9-4265-b8cc-6c814ebc7d90

Shall fix #468
closed 2024-10-16 by martinRenou