jupytercad/JupyterCAD #471: RTC is broken on the main branch

trungleduc, 2024-10-11:
With the latest changes in `jupyter-collaboration`, the RTC features do not work anymore, which causes failed UI tests.

SylvainCorlay, 2024-10-13:
cc @davidbrochart

SylvainCorlay, 2024-10-13:
It seems to be also preventing the JupyterLite deployment.

trungleduc, 2024-10-13:
> It seems to be also preventing the JupyterLite deployment.

Yes, although JupyterLite itself should work since it does not depend on jupyter-collaboration, we require UI test passing before deploying JupyterLite

davidbrochart, 2024-10-14:
I didn't look at the issue yet, but my first bet would be that jupyter-collaboration v3.0.0b8 has a new package `@jupyter/collaborative-drive` from which you import:
```ts
import {
  ICollaborativeDrive,
  ISharedModelFactory,
  SharedDocumentFactory
} from '@jupyter/collaborative-drive';
```
Instead of importing from `@jupyter/docprovider`.

PR #472 (2024-10-15): Update jupyter-collaboration v3.0.0-beta.8
Closes #471.

[yjs-widgets](https://github.com/QuantStack/yjs-widgets) will also have to use jupyter-collaboration v3.
closed 2024-10-15 by SylvainCorlay