jupytercad/JupyterCAD #478: Selecting edges used to be easier

martinRenou, 2024-10-16:
3D meshes used to have "sticky" edges, I recall this was achieved by having edges have a larger hitbox than their actual visual width.

It looks like it is not the case anymore (not sure since when?).

The result of this is that it is now more difficult to select edges as the hit needs to be pixel perfect.

martinRenou, 2024-10-16:
Digging out the original implementation, I see this: https://github.com/jupytercad/JupyterCAD/pull/340/files#diff-f0707db9d018c0ab205f1c1f88d4ecf5e9be247b731cbe2e7700726686a9e7e7R139

So maybe `this._raycaster.params.Line2.threshold = 50;` is not working anymore? I see there is still a threshold defined here `https://github.com/jupytercad/JupyterCAD/blob/8facc7c4ab419838e4e66f085e27622ecc854cfb/packages/base/src/3dview/mainview.tsx#L103` but it's for `Line` and not `Line2`.

Wild guess, is the unit of the threshold not the same for `Line` and `Line2`? I wonder if increasing the threshold fixes this?

arjxn-py, 2024-10-17:
> Digging out the original implementation, I see this: https://github.com/jupytercad/JupyterCAD/pull/340/files#diff-f0707db9d018c0ab205f1c1f88d4ecf5e9be247b731cbe2e7700726686a9e7e7R139
>
> So maybe [...]

It does the work, thanks @martinRenou 🚀

PR #498 (2024-10-17): Streamline edge selection
Shall fix #478 & #482 

https://github.com/user-attachments/assets/b9985faf-5095-43e1-8a63-1786cb84925a
closed 2024-10-17 by martinRenou