jupytercad/JupyterCAD #484: JupyterCAD icon should be highlighted in the file browser when the file is selected

SylvainCorlay, 2024-10-16:
I presume this would be achieved by adding the right CSS classes to the SVG files.

Screenshots below show the change os style of the selected file with a JCAD file vs a notebook.

[image] [image]

SylvainCorlay, 2024-10-16:
Related to https://github.com/jupytercad/JupyterCAD-FreeCAD/issues/14.

martinRenou, 2024-10-16:
We may not need to take any action on this if we change the design of the logo and the new logo fits nicer with the blue background.

Let's see if we need to take any action once that new logo is settled.

SylvainCorlay, 2024-10-16:
> We may not need to take any action on this if we change the design of the logo and the new logo fits nicer with the blue background.

It makes sense to wait for the new logo. Although even with the new logo, we should probably apply the selected style like for all other core Jupyter file types.

closed 2024-10-22 by martinRenou