jupytercad/JupyterCAD #526: After applying an operator the boundinx boxes of the inputs are still visible

martinRenou, 2024-10-22:
After applying an operator the boundinx boxes of the inputs are still visible:

https://github.com/user-attachments/assets/2126212d-9a65-4c9f-8d76-d7a8e9e10edc

I expected this not behave that way after putting the bounding box as part of the mesh group in https://github.com/jupytercad/JupyterCAD/pull/521, but it does not.

Maybe we're not setting the `visible` property on the whole group.

martinRenou, 2024-10-22:
Oh wait I guess it is the combination of two issues.

- I see the two inputs are still selected on the left panel: **the generated output should probably be selected after the operator is done instead**
- We should not create/show a bounding box for an object that's not visible

arjxn-py, 2024-10-22:
thanks for opening this @martinRenou, the explanation helps already and will look more into it

PR #555 (2024-10-30): Refactor Bounding Box Logic
Shall close #545 
Shall also close #526 

https://github.com/user-attachments/assets/207d755d-c66f-43e4-8135-2bf013e814ad
closed 2024-10-30 by trungleduc