jupytercad/JupyterCAD #535: STL and STEP file icons

SylvainCorlay, 2024-10-24:
It seems that the file icon is not showing up in the tab title for STL and STEP files.

On a separate note, it would probably make sense to not use the JupyterCAD logo for these file types, but use a different one for each.

trungleduc, 2024-10-24:
https://www.flaticon.com/free-icon/stl_9811308?term=stl+file&page=1&position=10&origin=search&related_id=9811308
https://www.flaticon.com/free-icon/stp_9811307?term=stp+file&page=1&position=3&origin=search&related_id=9811307

These icons look good

o314, 2024-10-30:
Here they are :

[image]

[image]

martinRenou, 2024-11-04:
Nice, thanks @o314! Did you make those yourself?

Only comment: the round corners are not the same for the underlying file icon and the filename label:

[image]

Otherwise it looks great!

SylvainCorlay, 2024-11-04:
@o314 @martinRenou thank you for your work on the icon.

I have a comment on the design: I think that it would look better with a sans-serif font for STL and STP.

o314, 2024-11-04:
OK. The font was not the good one since i have not found any probing setup in jupyter repo.
Of course, this has to be reworked - however this proto font was already sans serif.

Here is a file with variations on the font ; the formers are more cad oriented ; the later more commons.
If you 've got a jupyter specific one. point it up, i will brush up the icons

Last line is a more comprehensive rework, i finally prefer.
The current design tends to be overwhelmed with details . So i merge all backs layers into a unique sheet, common in cad
To reinforce the cad environment again, i've added a symbol that can be either understood as a drawing board or as a crane, so it encompasses the whole lifecycle of cad. The back layer is similar to the jupyter commons (same stroke, stroke width, dims) but not exactly due to the merge of back layers.

[image]

[jupytercad icons, curved.pdf](https://github.com/user-attachments/files/17623725/jupytercad.icons.curved.pdf)

The display is a rasterized shot, the file is a vectorized one.
The design file is not svg. Tell me what is your choice, i will provide u the corresponding svg file.

arjxn-py, 2024-11-05:
Super duper @o314, these looks awesome.
I like almost all of them. Hence don't have a strong opinion on any single design, tho i think the below two rows are not going to align well with the minimal design :)

I'll leave it up on others to choose any one. So Good :)

martinRenou, 2024-11-05:
Thanks a lot @o314 that's very much appreciated! Same opinion as @arjxn-py on this, they all look good and I also believe the last two rows don't match the rest of the JupyterLab design.

SylvainCorlay, 2024-11-06:
I agree with @martinRenou that the onces matchings the JupyterLab design would be a better choice. Regarding the choice of font, I would go for a thick one because the thin versions may not be very readable at small sizes.

o314, 2024-11-06:
ok. the 3rd and 4th columns are the declension for the bold font of the same family when available

martinRenou, 2024-11-08:
@o314 do you have an svg version of your icons handy?

o314, 2024-11-10:
Here are the ones that have been vectorized from fira mono bold

[image]

[image]

martinRenou, 2024-11-28:
This is fixed! Thanks @o314 for the icons, you've been credited for this :)

closed 2024-11-28 by martinRenou

o314, 2024-11-28:
You're welcome !
Thanks for your work and kind words too
*In english, from Lyon*

PR #591 (2024-11-28): Add stl and stp icons
Resolves #535