jupytercad/JupyterCAD #544: Cannot download JCAD file in JupyterCAD

SylvainCorlay, 2024-10-25:
Testing from JupyterLite, this results in the following error in the JS console.

```
Uncaught (in promise) Error: Content whose type is "jcad" cannot be downloaded
    at u (index.js:127:23)
    at async index.js:170:34
```

martinRenou, 2024-11-06:
It seems the (hidden) exception is `"Error: Content whose type is "jcad" cannot be downloaded"` coming from here https://github.com/jupyterlite/jupyterlite/blob/main/packages/application-extension/src/index.tsx#L209

I will see to fix it

martinRenou, 2024-11-06:
https://github.com/jupyterlite/jupyterlite/pull/1521 Should fix it

arjxn-py, 2024-11-06:
Thanks @martinRenou

SylvainCorlay, 2024-11-12:
Should we redeploy with JupyterLite v0.4.4 and close this issue @martinRenou ?

martinRenou, 2024-11-12:
Latest deployment uses jupyterlite-core 0.4.4 and I can confirm this is fixed and I can now download `.jcad` files after edits!

closed 2024-11-12 by martinRenou