jupytercad/JupyterCAD #550: Usage of commas as decimal points for some properties

SylvainCorlay, 2024-10-26:
I noticed that a comma was used for the decimal point for some properties of primitives (like position), while a dot was used for others, like for the radius of a sphere.

[image]

arjxn-py, 2024-10-26:
> I noticed that a comma was used for the decimal point for some properties of primitives (like position), while a dot was used for others, like for the radius of a sphere.

Strange, I can't even use commas in there

trungleduc, 2024-10-26:
me neither

SylvainCorlay, 2024-10-26:
Me neither, but the dot is replaced by a comma after submission.

SylvainCorlay, 2024-10-26:
**Note:** cComma is the standard decimal separator in France, but my operating system, keyboard, and all other settings are set to "English (US)".