jupytercad/JupyterCAD #557: Selecting objects is not smooth

arjxn-py, 2024-11-04:
Sometimes even after clicking on an object 2-3 times the objects are not selected.
Potentially related to https://github.com/jupytercad/JupyterCAD/pull/542

SylvainCorlay, 2024-11-04:
I think that the system often confuses a simple 'click' with a 'click and hold'.

arjxn-py, 2024-11-05:
> I think that the system often confuses a simple 'click' with a 'click and hold'.

Currently the very next click after panning is disabled, so when we want to select but it is also very likely that a very small panning/mouse movement has happened while or before clicking - it is leading one to click 2-3 or more times to select the object
We'll be on it how we can carve the logic to be smoother :)

martinRenou, 2024-11-05:
Yeah it looks like my PR https://github.com/jupytercad/JupyterCAD/pull/542 is not handling that correctly in some cases.

SylvainCorlay, 2024-11-05:
> Yeah it looks like my PR https://github.com/jupytercad/JupyterCAD/pull/542 is not handling that correctly in some cases.

It does not seem to be a question of time between the mousedown and mouseup, but rather than any (tiny) change of mouse position between when the mouse button is pressed and release prevents it to be considered as a click.

martinRenou, 2024-11-06:
> position between when the mouse button is pressed and release prevents it to be considered as a click

That's exactly what the implementation does in #542

martinRenou, 2024-11-06:
> Maybe we should count the number of pixels it has moved, and if it's less than a small amount of pixels (say 4), we consider it's a click

I'm not even convinced this is the only issue though.

There seems to be a wrong order of events that breaks the logic introduced in #542

arjxn-py, 2024-11-06:
Assigned this to myself to give a try (I might just be shooting in the dark at first 😅)

martinRenou, 2024-11-06:
Adding a note that the introduction of the transform controls for object now brings another point of complexity.

The transform controls deselects the selected object after transform. We should try to prevent that.

The mamba-jambo with all those events is starting to get very difficult to track this clicking behavior. We need to think of a robust approach for this.

PR #569 (2024-11-07): Fix selection logic
Finalize fix #557 

- Prevent triggering selection upon transform
- Fix non-reliable clicking logic where a single click would not always work
- Fix initial visibility of the bounding box highlight for selected objects
- Fix initial transform control position if an object is originally selected when the scene is loaded
closed 2024-11-07 by arjxn-py