jupytercad/JupyterCAD #585: Transform control icons

SylvainCorlay, 2024-11-22:
We should probably replace the placeholder for the transfor controls toggle icon before releasing.

[image]

arjxn-py, 2024-11-22:
Yes, once we have a decision on new Icon we shall be able to do it.

SylvainCorlay, 2024-11-22:
We could decide for a simple cross icon at first, that would be good enough for this version.

[image]

arjxn-py, 2024-11-22:
> We could decide for a simple cross icon at first, that would be good enough for this version.

Sounds good, i'm happy with this icon. Thanks :)

SylvainCorlay, 2024-11-22:
We should probably have our own drawing that we own for this repository.

PR #593 (2024-11-28): Add transform icon
Resolves #585