jupytercad/JupyterCAD #650: jupytercad not available from jupytercad's JupyterLite deployment

nthiery, 2024-12-10:
# How to reproduce

1. Click on the `try lite now` badge https://github.com/jupytercad/JupyterCAD
2. Create an empty Python3 notebook
3. Create a cell containing `import jupytercad`

This produces:
```
ModuleNotFoundError: No module named 'jupytercad'
```

Expected: jupytercad being imported to start having fun with it.

martinRenou, 2024-12-10:
Thanks for opening an issue! 

This is a known issue unfortunately, we can only have this once there is collaboration support in JupyterLite.

In JupyterLite only the JupyterCAD file viewer is available for now.

martinRenou, 2024-12-10:
Note that there was a package bringing collaboration support in JupyterLite https://github.com/jupyterlite/jupyterlab-webrtc-docprovider, we would need to revive this effort.

SylvainCorlay, 2024-12-10:
@martinRenou we also discussed using a service worker mimicking the websocket provider.

nthiery, 2024-12-10:
Oh, I see. It's fine. I just did not expect that when seeing a "try in lite now". Suggestion: add a brief README.md file which is open by default in JupyterLite, stating that the python API is not available yet there, but that one can click on the files to view them.

SylvainCorlay, 2025-04-24:
This appears to be resolved now. Closing.

closed 2025-04-24 by SylvainCorlay

nthiery, 2025-04-24:
I just retried and confirm. Thanks!