jupytercad/JupyterCAD #654: Do you have examples of JupyterCAD working in google colab?

willywong, 2024-12-13:
I am able to install JupyterCAD in colab, but unable to display/render any objects.

```py
#@title Create new CAD doc
from jupytercad import CadDocument

doc = CadDocument()

# Create a box, create a sphere, then cut the box using the sphere as a tool
doc = doc.add_box().add_sphere().cut()

display(doc)
```

This cell would return 

```
<jupytercad_lab.notebook.cad_document.CadDocument object at 0x7f721829c790>
```

martinRenou, 2024-12-14:
Unfortunately that cannot work in the current state of JupyterCAD.

I guess we could technically implement support for it by making a build of JupyterCAD that does not require jupyter collaboration to run in the front end

willywong, 2024-12-14:
Thanks for the quick reply. 

Google colab support would be a wonderful support, since it lowers the barrier of entry for sharing :)

Feel free to close as not planned/backlog/etc. Thanks.