jupytercad/JupyterCAD #672: Form "Ok" button is not using the --jp-brand-color1

martinRenou, 2025-01-21:
Sibling issue https://github.com/geojupyter/jupytergis/issues/375

vaishnavi-cdf, 2025-04-07:
Hi @martinRenou ,

I'm interested in working on this issue. I understand that the "Ok" button is currently not using the --jp-brand-color1 variable, which breaks consistency with JupyterLab's branding. I'd like to fix this and make sure the styling aligns with the design standards.

I'm actively contributing to open-source projects and looking for meaningful opportunities to learn and grow. If this issue is available, I would love to work on it. Please let me know if I can take it up or if any additional context is needed.

Thanks in advance for the opportunity!

martinRenou, 2025-04-07:
Thanks for your interest in fixing this!

If you look at the sibling issue in JupyterGIS and [the PR that fixed it](https://github.com/geojupyter/jupytergis/pull/379), you'll see what to do to port it here in JupyterCAD.

vaishnavi-cdf, 2025-04-07:
Thankyou