jupytercad/JupyterCAD #763: UsersItem toolbar not working anymore?

martinRenou, 2025-07-03:
I don't see the list of collaborators in the JCAD toolbar anymore, e.g. it's not possible to start the follow mode.

arjxn-py, 2025-07-03:
I remember when i was recording clips I faced the same with JupyterGIS so it might be a bug in `jupyter-collaboration` maybe?

brichet, 2025-07-11:
It seems to be an issue with the toolbar spacer that take all the space, like if it was not able to compute its size.
The spacer is supposed to fill the toolbar between left and right items, and I think that it needs the exact size of each item. 
If we remove the spacer DOM the users item is back.

[image]

martinRenou, 2025-07-15:
Yeah, the UsersItem is there in the DOM for me, just not visible because pushed outside of the parent component's area. So the spacer issue sounds like the culprit to me?

brichet, 2025-07-15:
It seems that the parent of the users item in DOM has no size, so the spacer doesn't take its content size into account, and fill the toolbar.
This seems to be because of the `position: absolute` of the[ `.jp-MenuBar-anonymousIcon`](https://github.com/jupyterlab/jupyter-collaboration/blob/499f8ff5df0dce6edaac7f79854292557fd12705/packages/collaboration/style/menu.css#L19), the default icon.

We should probably overwrite it, as it is done [here](https://github.com/jupytercad/JupyterCAD/blob/2a909068b191b9b89483d68fe0bf6fc9a0ec87b3/python/jupytercad_lab/style/base.css#L220), and think about fixing it upstream.

PR #767 (2025-07-15): Restore the users item visibility
Remove the `position: absolute` of the `.jp-MenuBar-anonymousIcon`, which led to a parent DOM with no size, no taken into account by the toolbar spacer.

Fix #763.
closed 2025-07-15 by martinRenou