jupytercad/JupyterCAD #774: Is it possible to group objects together?

asmith26, 2025-08-06:
Hi there,

Just wondering it is possible to group a load of objects together to make it easier to move them together?

Many thanks for any help! :)

arjxn-py, 2025-08-06:
Thanks for opening this issue, currently we do not have the ability to group objects and move them together unfortunately 

But it sounds like a useful feature to me and we would be happy to explore how we can implement this - for context we can already select multiple objects at once by clicking on them while pressing `Control` key - Maybe we can try integrating the combined movability of selected objects with the same?

asmith26, 2025-09-14:
Thanks for your help and info @arjxn-py. I had a thought: I tried the "Union/fuse" operator and it seems to do everything I need (e.g.more easily moving/rotating a group of objects), hence I'll close this.

Many thanks again! :)

closed 2025-09-14 by asmith26