jupytercad/JupyterCAD #777: "Export to .jcad" doesn't seem to be working

asmith26, 2025-08-21:
Hi there!

I've tried clicking the "File" > "Export to .jcad" button on both the "fan.stl" and "3M_CONNECTOR.STEP" examples ([JupyterLite](https://jupytercad.github.io/JupyterCAD) and running locally), and nothing seems to happen - I do seem to get a console error:

[image]

Many thanks for any help!

martinRenou, 2025-08-25:
Thanks for opening an issue.

Indeed, the export will not work in JupyterLite. We should hide the button and disable that feature in that case.

asmith26, 2025-08-25:
Thanks for your help and info @martinRenou 

In case helpful I also tried this locally and seemed to have the same problem.